francof2a/fxpmath #2: Problem with wrap saturation and numpy.float64

lpuglia, 2020-06-08:
hello,

I'm very new to this library but i think i found a bug related to the `+` operator:

This snippet works:
```
x = Fxp(7.25, False, 11, 8,overflow='saturate')
y = Fxp(7.25, False, 11, 8,overflow='saturate')
x.equal(x+y)
```
This one does not:
```
x = Fxp(7.25, False, 11, 8,overflow='wrap')
y = Fxp(7.25, False, 11, 8,overflow='saturate')
x.equal(x+y)
```
Erro stack:
```
~/.conda/envs/lpuglia_env/lib/python3.6/site-packages/fxpmath/objects.py in equal(self, x)
    362         if isinstance(x, Fxp):
    363             x = x()
--> 364         self.set_val(x)
    365         return self
    366 

~/.conda/envs/lpuglia_env/lib/python3.6/site-packages/fxpmath/objects.py in set_val(self, val, raw, vdtype)
    301         if val.dtype != complex:
    302             new_val = self._round(val * conv_factor , method=self.rounding)
--> 303             new_val = self._overflow_action(new_val, val_min, val_max)
    304             self.val = new_val.astype(val_dtype)
    305             self.real = self.get_val()

~/.conda/envs/lpuglia_env/lib/python3.6/site-packages/fxpmath/objects.py in _overflow_action(self, new_val, val_min, val_max)
    383             if new_val.ndim == 0:
    384                 if not ((new_val <= val_max) & (new_val >= val_min)):
--> 385                     val = utils.twos_complement_repr(new_val, self.n_word)
    386                 else:
    387                     val = new_val

~/.conda/envs/lpuglia_env/lib/python3.6/site-packages/fxpmath/utils.py in iterator(*args, **kwargs)
     48             return vals
     49         else:
---> 50             return func(*args, **kwargs)
     51     return iterator
     52 

~/.conda/envs/lpuglia_env/lib/python3.6/site-packages/fxpmath/utils.py in twos_complement_repr(val, nbits)
     58     else:
     59         val = val % (1 << nbits)
---> 60         if (val & (1 << (nbits - 1))) != 0:
     61             val = val - (1 << nbits)
     62     return val

TypeError: unsupported operand type(s) for &: 'numpy.float64' and 'int'
```

The problem seems related to the fact that 7.25 is internally represented using a numpy.float64, in fact the following code has no problem:

```
x = Fxp(7, False, 11, 8,overflow='wrap')
y = Fxp(7, False, 11, 8,overflow='saturate')
x.equal(x+y)
```
because both x and y are internally represented with a numpy.int64.

francof2a, 2020-06-08:
Hello,
Thank you very much for report the issue.
The problem is in twos_complement function, becuase it should cast val to int to use &.

I'va already correct that bug and update repo. I'm going to release version 0.3.3, so you will can update and this version should be works for you.

lpuglia, 2020-06-09:
Hey, thank you for the quick fix, it doesn't crash anymore but there are still problems:
```
print(Fxp(12.5, False, 11, 8, overflow='wrap'))
```
the output i get is:
```
7.205759403792794e+16
```
what do you think is the problem?

francof2a, 2020-06-09:
Hello,
Well, it's clear that _wrap overflow_ was not debug enough for unsigned Fxp. 
I found the error and it's fixed now. The wrap function was converting to signed during wrapping.
I tested the other routine you wrote, and it's ok also. I'm going to update repo and release with this fix.

Thanks for found it. Please, let me know when you test the new release to close the issue.