francof2a/fxpmath #6: Installation Failure on Clean Environment

jcharlong, 2020-06-19:
I am getting the following error when trying to install my package in CI when it goes to install fxpmath:
```
 Collecting fxpmath>=0.3.5
   Downloading fxpmath-0.3.5.tar.gz (15 kB)
     ERROR: Command errored out with exit status 1:
      command: /usr/local/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-tbrstae7/fxpmath/setup.py'"'"'; __file__='"'"'/tmp/pip-install-tbrstae7/fxpmath/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-bgv57lo5
          cwd: /tmp/pip-install-tbrstae7/fxpmath/
     Complete output (9 lines):
     Traceback (most recent call last):
       File "<string>", line 1, in <module>
       File "/tmp/pip-install-tbrstae7/fxpmath/setup.py", line 11, in <module>
         _version = __import__('fxpmath').__version__
       File "/tmp/pip-install-tbrstae7/fxpmath/fxpmath/__init__.py", line 6, in <module>
         from . import objects
       File "/tmp/pip-install-tbrstae7/fxpmath/fxpmath/objects.py", line 36, in <module>
         import numpy as np
     ModuleNotFoundError: No module named 'numpy'
     ----------------------------------------
 ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

Due to the way that it is currently retrieving its version information, numpy must be installed prior to installing fxpmath. pip does not have a proper dependency resolver so in my situation fxpmath is installed before numpy, leading the numpy import in "objects.py" to fail during the installation.

jcharlong, 2020-06-19:
See #7 for a PR.

francof2a, 2020-06-19:
hello,
You must to install _numpy_ before if you are using pip for install, because is a required package for *fxpmath*.

Other option is install using conda in a virtual environment. Conda will resolve your package dependencies.

jcharlong, 2020-06-19:
Hi,

I see that it is a requirement but using `pip` to install packages is very standard. The PR #7 should make it so that `pip` can do its job when running `pip install fxpmath` & install `numpy` if it is not already installed, lessening the burden of knowing what packages should be installed before running a `pip install` command.

My use case is a CI job which tests/packages a different project, in which I would like to use/depend on `fxpmath`. My CI environment starts out clean, without `numpy` installed.

francof2a, 2020-06-19:
Thanks for explanation. Sorry, but I answered without seeing merge request.
The solution seems good!
Thank you. I'm going to accept the merge request and update package in pip/conda soon.
Don't forget close the issue if you test and it's ok for you.

jcharlong, 2020-06-19:
No worries!

I tested locally & was able to install the package with no problems. I'll go ahead & close out the issue.

Thanks for this awesome package and your quick response!

closed 2020-06-19 by jcharlong