francof2a/fxpmath #9: array (numoy) operations support

MaenMallah, 2020-06-30:
Hello all,

Thanks a lot of this library, I was planning to implement something similar for a project of mine when I found this and it is very useful.

I am trying to use it with ndarrays from numpy and found few shortcomings with workarounds, I am not sure if there are better workarounds or tips for this:

* ndarray slicing: 
 It is great that the library can handle and uses numpy arrays in the backend. However, slicing can be problematic. E.g: I have a 2d array A of size [M,N] and use `A_fxp = Fxp(A)`. This works wonders and as I understand A is stored as ndarray inside the Fxp object. Now If I slice this as `C = A_fxp[0]` it generates an ndarray of size N. I thought this would generate a Fxp object with the same proprieties as A_fxp containing the vector of length N. My workaround: `C = Fxp(A_fxp[0], like=A_fxp)` which is not so bad for me but might cause issues when using functions with no control over slicing inside them. Is there a better way to do this currently?

* Updating the values of ndarray inside Fxp object:
Updating a value numpy ndarray inside of Fxp object does not accept Fxp object. in the same example above (A = [M,N] and use A_fxp = Fxp(A)). x = Fxp(4). A_fxp[0,0] =  x does not work. My Workaround: A_fxp[0,0] =  x(). Again, this is not bad but I think it would be better if it casts the Fxp object to allow the update. 

* Performace: 
E.g.: Using the same array A = [M,N],

```python
A_fxp = Fxp(A)
for i in range(M):
    for j in range(N):
         for x in range(10000): # a lot of iterations 
             A_fxp[j,j] = A_fxp[j,j] + some_variable
```
This is very slow. I am not sure what is going in the background. Does the object update (evaluate) the whole array to make sure rounding is correct for every time an update to one value takes place?

francof2a, 2020-07-01:
Hello MaenMallah,

First of all, thank you for the feedback.

**ndarray slicing**:
Yes, you're right. In the slicing that operation should return a Fxp object. It should be include that in next release.

**Updating the values of ndarray inside Fxp object**:
Yes, the cast should be supported here. In that case you should be carefull if `A` and `x` have different characteristics; for i.e in your expample format of `A_fxp` depends of values of `A` (best format was found), and format of `x` will be likely different. It should be also included in next release.

**Performance**:
Yes... very slow! I've checked it and when you change one or more values, all array is update and stored again. In that process, the rounding is performed again and overflow checked again. It's pretty sure there is a better way to do it. I'll try to figure it out.

francof2a, 2020-07-29:
Hello @MaenMallah ,
All the points now are supported in version 0.3.7.
I get a better performance in the slice settings.

Please check if the issue is solved for you.

MaenMallah, 2020-07-30:
Hello @francof2a,

Thanks a lot of addressing them this fast. I will test and let you know. Sorry I didn't answer earlier, I just noticed your early answer. Please keep the good work going.

MaenMallah, 2020-07-30:
Hello @francof2a,

When I tested slicing I got the following error: 

y[0]
Traceback (most recent call last):

  File "<ipython-input-28-5bfdbcf39e94>", line 1, in <module>
    y[0]

  File "C:\ProgramData\Anaconda3\lib\site-packages\fxpmath\objects.py", line 761, in __getitem__
    return Fxp(self.get_val()[index], like=self)

  File "C:\ProgramData\Anaconda3\lib\site-packages\fxpmath\objects.py", line 131, in __init__
    self.set_val(val)

  File "C:\ProgramData\Anaconda3\lib\site-packages\fxpmath\objects.py", line 330, in set_val
    val_max =  (1 << self.n_word) - 1

TypeError: unsupported operand type(s) for <<: 'int' and 'NoneType'

I tired initializing y with a list and numpy array and both threw errors.

francof2a, 2020-07-30:
Could you please paste whole code? I couldn't reproduce that error.

MaenMallah, 2020-08-03:
Hello @francof2a, I tried replicating this on a different machine and it worked fine. I am not sure what was wrong with my first setup. I will test again there and notify if this is working as expected.

francof2a, 2020-08-26:
Hello @MaenMallah ,
do you have any news about this issue? If not, please, let me close the issue.

closed 2020-09-09 by MaenMallah