francof2a/fxpmath #12: hex conversion

FredKellerman, 2020-07-31:
Hi, everything is looking good but I found a inconsistent behavior with .hex.  It drops the leading nibble(s) depending upon the value of a fractional number.

Technically this is not wrong if you read its value as a human would.  I am working with large fixed point numbers and push them through a 32-bit interface.  I was using .hex() to do that and but learned that the length varies.  While not impossible to work around, I found it just easier to use .bin() which is consistent in the converted string length.

I am doing something like below to push the raw bits through a hardware interface between 2 processing systems, a cpu and fpga:

f = Fxp(float128(1)/float128(3), True, 256, 253)
s = f.bin()  # or .hex()
n = int(s[0:32], base=2)  # [0:4], base=16 if hex
# Push value n through a 32-bit hardware fifo
n = int(s[32:64], base=2)
# Push next value n through a 32-bit hardware fifo
#  do the same for the remaining 32-bit words

# on the other side pull them out of fifo and reassemble into the original fixed point value, then do processing

francof2a, 2020-07-31:
Hi Fred,
All of this problem with `hex` it is because is replicating `hex` python function.
In the last commit I update that method to include padding by default. The repo is updated, but I`m not going to do a release yet. So, you can pull the repo and install from sources.

An alternative option is to use the next workaround:

```python
h = '0x{0:0{1}X}'.format(int(f.val), int(np.ceil(f.n_word / 4)))
```

That was I implemented to solve this problem.

FredKellerman, 2020-07-31:
Hi Franco,  Thanks for the quick response!  I'm using .bin() for my situation.  I will retest, I'm sure it is fixed, closing this.

closed 2020-07-31 by FredKellerman