francof2a/fxpmath #13: Dividing convenience vs traditional behavior

FredKellerman, 2020-08-03:
Hi,  I almost submitted this differently, some seems by design but some seems maybe incorrect.  The results can be very confusing, though it leaves the fixed point numbers unusable if doing division.  Basically when dividing a fixed point number, precision is being reduced so it will converge and approaches zero very quickly.

What I need for my context is for the fractional bits to increase just like when multiplying up to my baseline fractional bits.  I'm often using irrational numbers.  A partial work around is to multiply by the inverse but then I can only carry precision of inverse to the built in Python and Numpy floating point numbers.  For now though float128 is good enough.

But still there is something I don't expect with the shifting of the int bits also when dividing and the fractional bits as well.

If I do:

```python
from fxpmath import Fxp

a = Fxp(1, True, 128, 124)
a.info()
```

This reports:
dtype		=	fxp-s128/124
Value		=	1.0

But then do:
```python
a = a / 10000
a.info()
```
This reports:
dtype		=	fxp-s63/62
Value		=	0.0001

I'm not sure why the number of int bits has changed by a factor it has. Further iteration of a = a / 10000 after a few rounds you will get (which seems to do this around 64-bits):
dtype		=	fxp-s1/0
Value		=	0.0

Dividing by a fixed point number results in the same result but then another oddity (to me) is that the num int bits changed by the same amount even though my divisor had far more int bits to start with:
```python
a = Fxp(1, True, 128, 124)
a = a / Fxp(10000, True, 148, 124)
```
This produces:
dtype		=	fxp-s63/62
Value		=	0.0001
inaccuray	=	True

If I get really crazy and iterate this a handful of times:
```python
a = a / Fxp(.0001, True, 148, 124)
a.info()
```
I then see this:
/usr/local/lib/python3.6/dist-packages/fxpmath/objects.py:390: RuntimeWarning: invalid value encountered in true_divide
  if not np.equal(val, new_val/conv_factor).all() :

francof2a, 2020-08-04:
Hi Fred,
This behavior was coded with that intention, because division returns (a new Fxp with) the sizes according range and precision for the result.
I have no problem with the open issues, all of this helps to mature the code, so... thanks for the issue.

It's true that maybe is convenient has a parameter to define the minimum size after a lot of divisions or a division in a huge number. Also, it could be also convenient to have a parameter to force maintain size after division (I'm gonna to think about it a little).

By the way, when I coded this I had in mind this situation would be written as following:

```python
from fxpmath import Fxp

a = Fxp(1, True, 128, 124)
b = Fxp().like(a)

b.equal(a/10000)   # division result is stored in b (arbitrary sizes)
b.info()

# alternative to `equal` method:
b(a/10000)
``` 
Assignment to a Fxp should be done with `equal` method or `call` method.
Maybe this is a workaround for you in this case. I would like you tell me how would be the natural way for you.

About iterating this:

```python
a = a / Fxp(.0001, True, 148, 124)
a.info()
```

The division returns a Fxp with more bits until reach 64 bits (this is `_n_word_max`). This is a internal limit to "auto size" in new Fxp object. This is something to solve.
Try iterating this:

```python
# changing the limit for auto size
a = Fxp( a() / .0001, n_word_max=128) 
a.info()  

# or defining the final format
a = Fxp( a() / .0001, True, n_word=256, n_frac=0) 
a.info()
```

Both  are not the right solution, so... I have to solve this to use directly the division.

FredKellerman, 2020-08-04:
This division without removing precision would be a nice to have but multiplying by the reciprocal does provide a method to do this.  I wouldn't spend to much time on this unless others feel a need for it also.

You have however touched on another issue which I was/am going to create a separate issue for.  If you operate between Fxp types it is working great, no issues but if you try to construct a new one, even using an existing Fxp and need more than 64-bits of fractional precision it does not work.

francof2a, 2021-02-17:
Hi @FredKellerman,

Long time after... this functionality should be solved with version 0.4.0

Thank you!

closed 2021-02-17 by francof2a