francof2a/fxpmath #14: Initializing Fxp beyond 64-bits precision, is there a way?

FredKellerman, 2020-08-05:
Hi Franco,  There appears to be no way to initialize beyond 64-bits of precision. I've even tried to use binary strings to initialize and at first I thought it might work but it does not.  When I tried to enter the initialization as binary data it will only set 64-bits which start to fill from the right to the left but then they slide to the left if beyond 64-bits are passed in.  To the right of the sliding left are 0's.  The only other way that comes to mind would be to manually and awkwardly set the raw bits of the value of the obj.

I also tried to use Decimal from the decimal package but when I use it there is an exception going off because fxpmath/objects.py appear to treat it as Numpy and tries to use the .trunc() method.  This .trunc() method is likely at least one of them that is reducing the precision as well.

/usr/local/lib/python3.6/dist-packages/fxpmath/objects.py in _round(self, val, method)
    456             rval = np.fix(val)
    457         elif method == 'trunc':
--> 458             rval = np.trunc(val)
    459         elif method is None or method == '':
    460             rval = val

Even if Fxp adds support beyond 64-bit initialization, the next hurdle is that the largest precision I can think of is numpy.longdouble which is 80-bits.  I would like to be able to initialize and use full precision (in my case 128-bits, with someday 256-bits).  Because the initialization truncates and isn't carrying the full precision my iterative equations bias and end up at zero much quicker than I need.

Here are a few examples to reproduce the truncation to 64 bits:

```python
a = Fxp(numpy.longdouble(-1)/numpy.longdouble(3), True, 128, 125)
a.info()
a.hex()
d = Fxp('0b01011101010110101011101010101010101010101010101010101010101001010101010101010101001010101001010101010101010', True, n_word=128, n_frac=125, rounding='around')
d.bin()
from decimal import *
dd = Decimal('.333333333333333333333333333333333333333333')
f = Fxp(dd, True, 128,125) 
```

francof2a, 2020-08-26:
Hello Fred,
Sorry about delay, but almost 100% of my time is been consumed by other projects. I'll be work on these issues about long precision numbers in my gaps of time. I hope to solve it soon. I've replicated same wrong results that you noticed.

About Decimal, is right. This type (class) is not supported (yet) as input.

FredKellerman, 2020-08-26:
Hi Franco,  Please no apologies, many of us are doing things here in our spare time!

francof2a, 2021-02-17:
Hi @FredKellerman,

Long time after... this functionality should be solved with version 0.4.0

Thank you!

closed 2021-02-17 by francof2a