francof2a/fxpmath #15: Initializing Fxp with hex if hex contains 'b' init crashes

FredKellerman, 2020-08-07:
Hi Franco,  Sorry about this but I have found if a 'b' hex char appears the string appears to be interpreted as binary and then it will crash if the digits are not 0 or 1

You can reproduce it with the following:

```python
f = Fxp('0xb', True, 10, 4)
```

narendiran1996, 2020-08-25:
I too had similar issue.
Solved it by changing the following line in ~/.local/lib/python3.8/site-packages/fxpmath/utils.py in the `str2num` function.
`if base == 2 or 'b' in x:` to `if base == 2 or 'b' in x[:2]:`

Location in github sources [util.py](https://github.com/francof2a/fxpmath/blob/master/fxpmath/utils.py) - Line number 154.

francof2a, 2020-11-12:
Hello!
This issue is resolved now. Thank you!

closed 2020-11-13 by francof2a