francof2a/fxpmath #19: Weird behavior of complex arrays math

arnfol, 2020-11-10:
Hi! 
I've faced some really strange behavior of sum of Fxp complex array elements. Here is an example:
```
from fxpmath import Fxp
import numpy as np

DW=12
DATA_FXPTYPE = Fxp(None, signed=True, n_word=DW, n_frac=DW-1)

a = Fxp(np.zeros(2, dtype=complex), like=DATA_FXPTYPE)
c = Fxp(complex(0,0), like=DATA_FXPTYPE)
b = Fxp(0.1-0.1j, like=DATA_FXPTYPE)

print(c.get_val() + b.get_val())
c.equal(c+b)
c.info()

print(a[0].get_val() + b.get_val())
a[0].equal(a[0]+b)
a[0].info()
```
I've got the following output:
```
(0.099609375-0.099609375j)
	dtype		=	fxp-s12/11-complex
	Value		=	(0.099609375-0.099609375j)

(0.099609375-0.099609375j)
	dtype		=	fxp-s12/11-complex
	Value		=	0j
```
So, for some reason it does work correctly then I am trying to sum single Fxp elements, but if done with Fxp array element something goes wrong, and the value is not updated.

francof2a, 2020-11-11:
Hello!
Thank you very much for the issue. I found a bug that only imaginary part is being stored, the fix will be included in the next release. I'll notify you.

Beside this fix, in your example, 
```python
a[0].equal(a[0]+b)
```
should be replaced by
```python
a[0] = a[0]+b
```

The reason is that `equal` doesn't support indexing.

arnfol, 2020-11-11:
Thank you for the quick response :) 

If I change `a[0].equal(a[0]+b)` to the assignement `a[0] = a[0]+b` in my example, I also have a problem:
```
(0.099609375-0.099609375j)
	dtype		=	fxp-s12/11-complex
	Value		=	(-0.099609375+0j)
```
As you can see the imaginary part somehow turned to zero(

By the way, do you have any thoughts about adding more array operations support? I found it difficult to convert my algorithm from classic numpy complex128 ndarrays to Fxp :D Maybe it is possible to create a custom numpy ndarray type for Fxp?

francof2a, 2020-11-11:
yes, you have that problem because I've not released the next version yet, sorry. I'm going to include the fix for your issue #17 to release it.

---

I'm aware about the difficult of conversion from numpy to fxpmath. It's not easy to do that. I think a good solution is in the path of  using a numpy **dispatcher**, [https://numpy.org/doc/stable/user/basics.dispatch.html](url). It is something really interesting to working on, maybe in near future, depending of my time, or maybe a collaborator :)

francof2a, 2020-11-12:
This issue is solved in new version (0.3.8).
Please, tell me if it's ok for you.

closed 2020-11-23 by francof2a