francof2a/fxpmath #21: Wrong shape in numpy.inner()

masan4444, 2020-12-07:
# Wrong shape in numpy.inner()
Hello,

Thanks a lot of this library

```
import numpy as np
from fxpmath import Fxp

a = [1, 2, 3]
b = [0, 1, 0]
np.inner(a, b)
# >>> 2

a = np.array([1, 2, 3])
b = np.array([0, 1, 0])
np.inner(a, b)
# >>> 2

a = Fxp([1, 2, 3])
b = Fxp([0, 1, 0])
np.inner(a, b)
# >>> [0 2 0]
```

## environment
Python 3.6.9
numpy 1.19.4
fxpmath 0.3.8

francof2a, 2020-12-09:
Hello @masan4444 
You're welcome!

I'm sorry about bug, but Fxp inputs for numpy's functions is not supported entirely yet. I'm working in the dispatch functionality for numpy's functions in *fxpmath* version 0.4.

By now, I can say that a work around would be:

```python
x = np.inner(a(), b())    # output like numpy ndarray
x = Fxp(np.inner(a(), b()))   # output like Fxp (non fixed-point calculation)
```

Doing `a()`, the val of `Fxp` is returned like a non fixed-point value. So, depending of your problem, maybe this work around is not useful.

francof2a, 2020-12-10:
I've test your code in the development branch [*dev_version_0_4*](https://github.com/francof2a/fxpmath/tree/dev_version_0_4) version **0.4.0-alpha.5**, and it is working right!

There is not a release of this, but you can clone and install it from the repo if you need.

francof2a, 2021-02-17:
This issue is solved with version 0.4.0.

Thank you!

closed 2021-02-17 by francof2a