francof2a/fxpmath #23: Python OverFlow errors. How to use?

JoachimKoenigslieb, 2021-01-14:
I'm trying to use this library to implement various algorithms for use in hardware systems. Running the calculations, I get errors like:
`OverflowError: int too large to convert to float`

This happens when the .val method (actual data representation) gets very large I think. Is there any workarounds or hints to what I'm doing wrong? I was expecting the overflow methods of the library to trigger such that I would never be able to make an illegal operation with elementary math functions.

francof2a, 2021-01-15:
Hello @JoachimKoenigslieb
The problem is related with numpy `int` representation, where 64 bits is the largest. I'm actually working in a new version (0.4) where that problem is fixed. By now, it is not available in pip, just in the branch [dev_version_0_4](https://github.com/francof2a/fxpmath/tree/dev_version_0_4).

You may clone it and try, or just comment a piece of code here in the issue and I'll test it.

JoachimKoenigslieb, 2021-01-15:
Ok. I will take a look at this branch! Im planning on using this library for testing implentations of fluid dynamics on fixed point hardware. I might write some PRs for enabling fast and efficient profiling on numpy codes. Would you be open for this?

closed 2021-01-15 by JoachimKoenigslieb

francof2a, 2021-01-18:
Very interesting project!

is PRs for _Pull Requests_? (Just to confirm)
I have no problem about that! I try to keep an eye on performance, but by now it is not my main concern, but functionality. It is very likely that many things can be improved in terms of performance.

JoachimKoenigslieb, 2021-01-19:
PR Is pull request! By "fast and efficient" I meant from a code-useability
point of view! Ie, how to easily integrate into existing numpy projects!
Over the next few days I might write you a few questions related to the
(very nicely commented) codebase. Hopefuly you will have time for some
short answers.