francof2a/fxpmath #26: How to convert an array of strings representing hex?

ricardo-jimenez, 2021-03-01:
Does this library support vectorized operations using arrays of strings as inputs?  how?
I am trying to implement something like what you have in the example notebook for [ADC_sampling_quantization](https://github.com/francof2a/fxpmath/blob/master/examples/ADC_sampling_quantization.ipynb) but with the signal being a numpy array of strings. 

For instance using: 
`sig = np.array(['0xff864d8f', '0xff86b76d', '0xff880f87'])`  
in 
`Fxp(sig)` 

I get the following error
[image]

francof2a, 2021-03-02:
Hi @ricardo-jimenez 
Thank you for reporting this problem.
I was able to reproduce the problem and it is in the best size estimation function.

A workaround is using `tolist` method to pass a list of strings and define the size for the Fxp object:

```python
Fxp(sig.tolist(), signed=False, n_word=32, n_frac=0)
```

I'll be working to solve this!

francof2a, 2021-07-05:
fixed at v0.4.1

closed 2021-07-05 by francof2a