francof2a/fxpmath #27: Incomplete/inconsistent documentation of rounding methods

hcommin, 2021-03-03:
The _trunc_ rounding method is documented as follows:

> trunc (default): The truncated value of the scalar (let's say x) will be the nearest fractional supported value which is closer to zero than x is. In short, the fractional part of the signed number x that is not supported, is discarded.

If the value is signed, then rounding towards zero is not the same as discarding the fractional part. If this is actually a truncation, then in general it rounds towards negative infinity, not zero.

It seems like at least two of _trunc_, _floor_ and _fix_ must be doing the same thing, but I can't determine which because the descriptions are very unclear.

The _around_ rounding method is documented as follows:

> around : Evenly round of the given value to the nearest fractional supported value.

The problem here is that you haven't defined what is considered "nearest" in the case of a tie. (For example, is 1.5 closer to 1.0 or 2.0?) There are at least 4 common definitions, each of which may be useful in different contexts (half-up, half-down, convergent-even and convergent-odd).

francof2a, 2021-03-04:
Rounding is applied before fixed-point conversion. In documentation:

> That is because Fxp is rounding the value before storing as a fractional fixed point value. 

So, rounding function is applied to x ** (1 << n_frac); where `x` is float value. All the rounding functions are from *Numpy*, and each description in *fxpmath* documentation is according *Numpy* documentation. For example: [numpy.trunc](https://numpy.org/doc/stable/reference/generated/numpy.trunc.html).

---

> If the value is signed, then rounding towards zero is not the same as discarding the fractional part. If this is actually a truncation, then in general it rounds towards negative infinity, not zero.

It's not a binary truncation performed, see above.

---

> It seems like at least two of trunc, floor and fix must be doing the same thing, but I can't determine which because the descriptions are very unclear.

`fix` and `trunc` do the same in Numpy also:

```python
w = np.random.uniform(-16, 16, size=(2000,))
np.all(np.trunc(w) == np.fix(w))
```

> True

---

> The problem here is that you haven't defined what is considered "nearest" in the case of a tie. (For example, is 1.5 closer to 1.0 or 2.0?) There are at least 4 common definitions, each of which may be useful in different contexts (half-up, half-down, convergent-even and convergent-odd).

The float value scaled is **evenly rounded**, [wikipedia Round half to even](https://en.wikipedia.org/wiki/Rounding#Round_half_to_even). So, 1.5 is rounded to 2.0.

[image]

[image]

francof2a, 2021-05-31:
Documentation updated!

closed 2021-05-31 by francof2a