francof2a/fxpmath #28: Incorrect behavior of basic arithmetic operations

hcommin, 2021-03-03:
According to the documentation:

> This math operations returns a new Fxp object with automatic precision enough to represent the result. So, in all these cases we can assign the result to a (Fxp) variable, or to the same (overwritting the old Fxp object).

However, this is clearly not working at all. Taking some examples directly from the documentation:
```
x = Fxp(-7.25)
y = x**3
z = 3.25 * (x - 0.5)
```

This results in y = z = -8.0.

francof2a, 2021-03-03:
hello @hcommin,
Thank you for this report.

There are some (special) functionalities that you should be aware of here. One of them is when you do a math operation like `Fxp` (.) `Constant`, the resulting Fxp object has the same size as the Fxp used as operand.

```python
x = Fxp(-7.25)
x.info(3)
```
> ...
>	Upper		=	7.75
>	Lower		=	-8.0
> ...

You can see that (-7.25)**3 = -381.078125 < -8.0, and that's because the value of `y` is -8.0.

Depending of what you're trying to modeling, there are different ways to procedure:
```python
x = Fxp(-7.25, dtype='fxp-s16/6') # fixed point space is enough to store the result
y = x ** 3
y
```

> fxp-s16/6(-381.078125)

If you want that resulting Fxp has `optimal` size you could do:

```python
x = Fxp(-7.25)
y = x ** Fxp(3) # 3 is converted to Fxp with optimal size before pow operation
y
```
> fxp-s17/6(-381.078125)

or change the _behavior_ of `x` when operates with constants:

```python
x = Fxp(-7.25)
x.config.const_op_sizing = 'optimal'  # default value is 'same'
y = x ** 3
y
```

> fxp-s16/6(-381.078125)

You can find more details about behaviors in this example [v0_4_0_new_functionalities.ipynb](https://github.com/francof2a/fxpmath/blob/master/examples/v0_4_0_new_functionalities.ipynb)

hcommin, 2021-03-04:
@francof2a If there is any functionality that I should be aware of, then it should be documented. None of what you wrote addresses the fact that the documentation is wrong:

> This math operations returns a new Fxp object with automatic precision enough to represent the result.

francof2a, 2021-03-04:
There's a lot of things that remains undocumented. Be aware is a beta release (0.x.x).
The name of issue es about behavior and not about documentation. So my answer was wrote in that way. I consider that there's not problem about behavior, so I'm going to close the issue.

I thank you, by the way, about notice that in documentation (it is not updated).

closed 2021-03-04 by francof2a