francof2a/fxpmath #29: Reinterpret fixed-point number

udara28, 2021-03-11:
Is there a way I can reinterpret a fixed point number with different number of fractional bits? 

In my use case I'm receiving MyHDL intbv objects which are actually fixed-point values with some fractional bits.  If I create a Fxp object like Fxp(int(x), True, 5, 2) then it will modify the bit pattern. For example say I have 2 fractional bits. if x=0b00010 then x actually represent 0.5.  But if I do Fxp(int(x), True,  5, 2) it will become x=0b01000. Is there a way I can keep x as 0b0010 and just change the number of fractional bits so that it can be interpreted as 0.5 or 2?

francof2a, 2021-03-15:
Hello @udara28 

I think you need to load a _raw_ value in _Fxp_, or I misunderstood your problem.

If I was right, you can do that as following:

```python
x_fxp = Fxp(int(x), True, 5, 2, raw=True)
```
or you can assign new raw value after creation:

```python
# x_fxp = Fxp(...)
# ...
x_fxp.set_val(int(x), raw=True)
```

Please, tell me if it solved in this way

udara28, 2021-03-15:
@francof2a thanks for your reply. I think the first method can't work since raw is not a parameter in the Fxp init function. The second method did the trick for me. I didn't notice any mention about the raw parameter in the README. So it would be really helpful if it can be added there. Thanks again for the help and I'm closing this issue.

closed 2021-03-15 by udara28

francof2a, 2021-03-16:
The first method is going to work because `raw` is passed as _kwarg_ to the init function.

You're right about documentation. It is not updated. I hope update soon. Thank you!