francof2a/fxpmath #30: seems the resize() method has incorrrect output

steven918, 2021-04-03:
The resize function itself seems correct but maybe some overflow is not defined right. Use the ADC quantization example, do the following:

... setup f_sig, n, n_samples as shown in the example, 

sig = A*np.sin(2*np.pi*f_sig*n/n_samples)
sig_8_7=Fxp(sig, signed=True, n_word=8, n_frac=7)
sig_8_4=Fxp(sig, signed=True, n_word=8, n_frac=4)
sig_8_4[21].resize(True, 8, 7)        # ==> just pick a single point
sig_8_4[21].bin(frac_dot=True)    #  ==> sig_8_4[21] is not resized to the same size as sig_8_7[21]
sig_8_7[21].resize(True, 8, 4)
sig_8_7[21].bin(frac_dot=True)    # ==> sig_8_7[21] is not resized to the same as sig_8_4[21] either

Another example that had incorrect resize output too:
 x = Fxp(10.75, dtype='S5.16')
x.bin(frac_dot=True)        # return '01010.1100000000000000'
x.resize(True, n_word=12, n_frac=9, restore_val=True)
x.bin(frac_dot=True)        # return '011.111111111' or 3.99. Correct answer should be 10.75/2**7=0.083984375

francof2a, 2021-04-08:
Hello @steven918 ,

**First case**:
The `resize` method is not allowed to each element, because a Fxp object is homogeneous in size. So, I think you need to work with a list of Fxp objects instead a Fxp of a list (or numpy array) if you want to change the size of individual elements.

**Second case**:
`restore_val` is True by default and it does that original value should be keep in new size, but resized `Fxp` object has to be able to represent that value. In your example, resized `x` has upper limit of 3.998046875 because of 2 bit for integer part and 9 bits for fractional. The original value of 10.75 is truncated to that upper limit.
I don't understand what you wrote as correct answer.

steven918, 2021-04-08:
Thanks @francof2a,

I understand your answer to the first case where a list was used now. 

As for the second case, I thought "restore_val" is intended to have the same value after resizing. But it seems setting "restore_val" to False still returned the same value as it is set to True. Both of them returned 

x.bin(frac_dot=True)
Out[11]: '011.111111111'

My understanding of the resize operator was scaled version of the input: before scaling there were 16-bit fractional and after scaling there was 9-bit fractional so from fixed point representation "real_value = scale * x" (assuming zero biased and x is the fixed point), the resized value would be shifted by (16-9=7) bits. Of course, this is not intended to be a "restore_val" operation and it sounds not supported by the resize() function

francof2a, 2021-04-09:
if `restore_val` is False, the resized Fxp is going to keep de raw value (fixed point, not fractional), but for that Fxp resized must have enough word *bits) size to store that value. In the second case, you are resizing from a 21 bits Fxp to a 12 bits Fxp, that's because truncation occurs also.

if you do:

```python
x.resize(True, n_word=12, n_frac=9, restore_val=False) # keeping same word size
x.bin(frac_dot=True)
```
> Out[ ]: '010101100000.000000000'

You have same binary but with different fractional size. That's the function of `resize` method, but it's not rescaling function you need.

If I understood well what you need, maybe following is a solution:

```python
x = x >> 7
x.resize(True, n_word=12, n_frac=9, restore_val=True)
print(x)
print(x.bin(frac_dot=True))
```
> 0.083984375
> 000.000101011

Please, let me know if this is what you're trying to implement.

steven918, 2021-04-09:
oh, I see. You intended not to keep the fractional value after resizing. You're right that the scaling version is simply a right shift of input and then resize. This is the right shift and decimal alignment (for example through resize()) process I was looking for and thought the resize() would do. Thanks for clarify that.

I was trying to find a similar fxp version to match to Tensorflow's quantization (in tf.quantization.quantize), which quantize float to qint8, qint16, etc. That's why I bumped into your implementation. Thanks for all the answers and great work!

francof2a, 2021-04-09:
You are welcome. Thanks for using this module and report your problem.
I think your issue is solved. I'm going to close, but if you consider that something remains, please don't doubt to open a new one.

P.S.: It would be great that **fxpmath** could work with tensorflow!

closed 2021-04-09 by francof2a

steven918, 2021-04-09:
Yes, please close it and label "solved". I will certainly bug you if I run into something else.