francof2a/fxpmath #31: Internal numpy array incorrectly stores truncated values

DanChianucci, 2021-05-12:
It looks like the internally stored np array doesn't always use the correct dtype.

From the example below you can see that depending on how the variable are initialized, certain operations don't work.

Given the fxp data type, I would have expected all internal arrays to be of uint64, especially since they are initialized from the same "constructor"

```python

from fxpmath import Fxp,Config

t= Fxp(2**32,dtype="u32.32",shifting="trunc")
print(t.dtype , t.val.dtype ,  t )    # fxp-u64/32 int32 0.0

s  = t(0.125);
print(s.dtype , s.val.dtype ,  s )    # fxp-u64/32 int32 0.125

q  = t(0.125)<<3;
print(q.dtype , q.val.dtype ,  q )    # fxp-u64/32 int32 0.0

q2 = t(0.125)*2**3;
print(q2.dtype, q2.val.dtype,  q2)    # fxp-u64/32 int32 0.0

q3 = t(0.125*2**3);
print(q3.dtype, q3.val.dtype,  q3)    # fxp-u64/32 int64 1.0

# Looks like the above q3 statement upconverted 
# the backend type, and in doing so modified how 
# s is interpretted even though no changes are made 
# to s
print(s.dtype , s.val.dtype ,  s )    #  fxp-u64/32 int64 1.0
```

francof2a, 2021-05-13:
Hello @DanChianucci 
You're right! There was a problem with casting to an `object` inside constructor and inside `multiply`. The problem is because Numpy doesn't support (implement) that automatic type conversion. In your example you are working with a raw value equal to 2**64, that is not supported by `uint64`.

The only behavior I could not reproduce is that `val.dtype` is `int32` instead of `int64`. In my PC `int64` was returned. I suppouse that is because your Numpy version or your operative system. Could you tell me which is your Numpy version (pip and/or conda) and your OS?

In the last commit the issue would be solved! If it is ok it will be released soon with v0.4.1.

The print outputs for your example are next:

```
In [2]: t= Fxp(2**32,dtype="u32.32",shifting="trunc")
   ...: print(t.dtype , t.val.dtype ,  t )
fxp-u64/32 object 4294967296.0

In [3]: s  = t(0.125);
   ...: print(s.dtype , s.val.dtype ,  s )
fxp-u64/32 object 0.125

In [4]: q  = t(0.125)<<3;
   ...: print(q.dtype , q.val.dtype ,  q )
fxp-u64/32 object 1.0

In [5]: q2 = t(0.125)*2**3;
   ...: print(q2.dtype, q2.val.dtype,  q2)
fxp-u64/32 object 1.0

In [6]: q3 = t(0.125*2**3);
   ...: print(q3.dtype, q3.val.dtype,  q3)
fxp-u64/32 object 1.0

In [7]: print(s.dtype , s.val.dtype ,  s )
fxp-u64/32 object 1.0
```

DanChianucci, 2021-05-13:
I'm using 
python:  3.9.1
conda:   4.10.1
Numpy: 1.19.2

francof2a, 2021-05-14:
I had no problem with those versions.
You can try:

```python
import numpy as np

np.array(2**64-1).astype(np.uint)
```
 
Check if `dtype` returned is `np.uint32` or `np.uint64`. Last one should be returned, except your numpy is running in 32 bits mode. That would explain the behavior.

In both cases, the solution committed works.

DanChianucci, 2021-05-17:
That explains it,  my setup is returning dtype as uint32.   

I did try your fix and can confirm it is now working as expected with this setup

francof2a, 2021-07-05:
fixed at v0.4.1

closed 2021-07-05 by francof2a