francof2a/fxpmath #32: Rounding not applied properly when instantiating on ndarray of values that will overflow

nfrancque, 2021-05-23:
I have an ndarray that overflowed from another operation to -7.70371978e-34 and upcasted itself to another dtype due to some arithmetic

If I then call Fxp(...) on this object I would expect the new value to be set to the min of whatever dtype is chosen. It looks like it tries to do this but this last step clear everything to zero (line ~600 of objects.py)
```python
            if val_dtype == object:       
                # convert each element to int
                new_val = np.array(list(map(int, new_val.flatten()))).reshape(new_val.shape)
```

If I rewrite this as the following it seems to fix the issue but haven't tested thoroughly beyond that it seems to have the proper rounded values.

```python
            if val_dtype == object:       
                # convert each element to int
                new_val = np.array(list(new_val.flatten())).reshape(new_val.shape)
```

Is there a reason they all need to be mapped to integers?  Seems like it should be something else.

francof2a, 2021-05-26:
Thanks for the issue!
It would be good if you could write an example to reproduce the error.

The integer mapping is because Numpy. If the `Fxp` has more than 64 bits, Numpy rise error in some cases because can't allocate more than 64 bits for `np.int` or (more dangerous) use `float` and rounds values using that precision. For that reason, it's necessary to use `int` python objects in those cases (`object` dtype for Numpy).

Maybe something useful for you is **callbacks**, to call a function when an _overflow_ or _underflow_ occurs. You can find and example in [examples/v0_4_0_new_functionalities.ipynb](https://github.com/francof2a/fxpmath/blob/master/examples/v0_4_0_new_functionalities.ipynb). I'm going to copy part of it next:

```python
from fxpmath.callbacks import Callback

def print_over_underflow_msg(self):
    print('Something is going wrong, some value is outside limits for this Fxp!')

my_callback = Callback()
my_callback.on_status_overflow = print_over_underflow_msg
my_callback.on_status_underflow = print_over_underflow_msg

x = Fxp(dtype='fxp-s16/4', callbacks=[my_callback])
```

francof2a, 2021-08-05:
Closing the issue due lack of answer. Feel free to reopen it if you consider it necessary.

closed 2021-08-05 by francof2a

nfrancque, 2021-12-22:
Hey @francof2a , just wanted to say thanks for pointing out the callbacks that is pretty much what I was looking for - just a global exception configurable to a warning to be thrown on overflow.

This might be worth adding into the module, or for someone reading this thread later.  I have wrapper functions like this:
```python
def gen_fxp_handler(type):
    def msg_printer(self):
        pprint(self.__dict__)
        assert False, f"Fxp {type}"
    return msg_printer

fxp_callback = Callback()
fxp_callback.on_status_overflow = gen_fxp_handler("overflow")
fxp_callback.on_status_underflow = gen_fxp_handler("underflow")

def to_fixed(float_num, total_bits, fraction_bits):
    return Fxp(float_num, signed=True, n_word=total_bits, n_frac=fraction_bits, callbacks=[fxp_callback],
        rounding='trunc', overflow='saturate')

def to_float(fixed_num):
    ret = fixed_num.get_val()
    ret = ret.astype(np.float64)
    return ret
```

this way it gets easier to keep the same parameters to the whole application and not have to know there's an Fxp object running around.  IMO it'd be nice if there was some global config that could be applied to Fxp in the current scope for configuring whether overflow/underflow is an error/warning/nothing in addition to the other parameters.

This is a very useful library, thanks!

francof2a, 2021-12-22:
Hi @nfrancque ,
Thanks for the code snippet! 
It's glad to hear about **fxpmath** is useful for you.

The automatic **error/warning/nothing** behavior is a very good idea. It's pretty sure that I'll include it in next release.
Thanks!