francof2a/fxpmath #39: python warning

mehdimolum, 2021-09-06:
> Hi, 
I am using python 3.8.10 and get following error

`/home/matt/.local/lib/python3.8/site-packages/fxpmath/objects.py:597: FutureWarning: Conversion of the second argument of issubdtype from `complex` to `np.complexfloating` is deprecated. In future, it will be treated as `np.complex128 == np.dtype(complex).type`.
  if original_vdtype != complex and not np.issubdtype(original_vdtype, complex):` 

That is only  awarning and not an issue for my work but that would be good to fix it when got time.

BWT, thanks for the library. helped with my project a lot!

francof2a, 2021-09-07:
Hello @mehdimolum 

Thanks for reporting the warning. It will be taken into account in next release!

I'm very glad about the library is useful for your project.

francof2a, 2021-09-21:
It should be solved in version 0.4.3.
I'm going to close the issue. Feel free to reopen it if the problem was not solved.
Thanks!

closed 2021-09-21 by francof2a