francof2a/fxpmath #40: How can I convert a signed raw Fxp representation back to a float?

aniongithub, 2021-09-08:
I need to convert Fxp processed raw, signed float representations back to floats in (say) a CUDA kernel (I can't use Fxp for this reverse conversion). For unsigned representations,

`floatVal = rawVal / (2 ^ 8)`

works fine and matches up with the expected value. However, I see no information on how Fxp deals with signed values. How can I convert a raw representation of a signed float back into a signed float?

Thanks!

aniongithub, 2021-09-09:
I figured it out - leaving a link to the solution here for anyone else that might want to do the same. Since I'm using cupy, I wrote this for numpy so the answer is broadly applicable.

Gist with code: https://gist.github.com/aniongithub/50050fc192ac1a16e144bc6de51252db

closed 2021-09-09 by aniongithub

francof2a, 2021-09-09:
Hello @aniongithub 
It's great that you found the way to figure it out! You're right about calculate the two's complement before conversion.

Let me left some code by here to resolve that procedure using _fxpmath_:

```python
from fxpmath import Fxp
from fxpmath.utils import twos_complement_repr

float_vals = [1.25, -2.5]
x = Fxp(float_vals, signed=True, n_word=32, n_frac=8)
uraw_vals = x.uraw() # get the unsigned raw value, you could cast to uint32 if you want doing x.uraw().astype(np.uint32)

# to recover the float value you could do:
r = Fxp(twos_complement_repr(uraw_vals, x.n_word), raw=True, like=x)
# you can change `x.n_word` to amount of bits of raw binary representation

# checking both are the same
assert (x == r).all()
```