francof2a/fxpmath #41: Overflow behaviour at instantiation fails for long words

desvdp, 2021-09-17:
It seems the overflow setting fails at instantiation when using wordlengths over 64bit

Fxp(2,False,63,0,overflow='wrap') --> Ok
Fxp(2,False,64,0,overflow='wrap') --> Throws error

francof2a, 2021-09-20:
Hello @desvdp,

Thanks for the issue. I was able to reproduce the error. I'll debug this soon!
Same for the issue #42

francof2a, 2021-09-21:
It should be solved in version 0.4.3.
I'm going to close the issue. Feel free to reopen it if the problem was not solved.
Thanks!

francof2a, 2022-05-13:
The last code line in error report (_fxpmath\utils.py, line 352_) should be:

```python
x = int_array(x).astype(dtype) & (m - 1)
```

I change this in `v0.4.9-dev0` (branch **dev-0.4.9**). 
This change should fix the problem. I'll include it in next release.

Let me know if you test this solution.
Thanks!

alb-pol, 2022-05-13:
What about line 356 in fxpmath\utils.py, also int_array as in line 352?

francof2a, 2022-05-13:
you're right, this change is valid for unsigned values also. I added this and make little changes to this function.