francof2a/fxpmath #43: .bz2 for offline installation?

Phillip-M-Feldman, 2021-09-20:
Is there a .bz2 version of the package for offline installation?

francof2a, 2021-09-21:
No, sorry.
You can download `tar.gz` or `.zip` from [https://github.com/francof2a/fxpmath/releases](https://github.com/francof2a/fxpmath/releases).

If you're in Linux, you can download `tar.gz` file and try in console:

```
gunzip < fxpmath-0.4.3.tar.gz | bzip2 > fxpmath-0.4.3.tar.bz2
```
if you want to convert format.

---

[Conda distribution files ](https://anaconda.org/francof2a/fxpmath/files) are compressed  as `tar.bz2`. You can find source code for library inside (_/lib/pythonX.X/site-packages/fxpmath/_), but `setup.py` is not included in this case.

Phillip-M-Feldman, 2021-09-22:
Thanks!  This is a great package.

Phillip

francof2a, 2021-09-22:
Thank you Phillip. I hope this package was useful for you.

I'm going to close the issue. Feel free to reopen it if you consider that it isn't solved.

closed 2021-09-22 by francof2a