francof2a/fxpmath #49: Reversal of .bin()

nfrancque, 2021-12-22:
I'm wondering if there is a way to revert calling `.bin()`.  Something like:

```python
x1 = Fxp(3.4)
x_bin = x1.bin()
x2 = from_bin(x_bin)
assert x1 == x2
```

My use case is to convert into the notation, do some operations, then convert the result back over.  I see the constructor can take in a binary value but it looks like it only takes in a "normal" python bit string.  Is this possible currently?

francof2a, 2021-12-22:
The method `bin` returns a string without `0b` prefix to maintain compatibility with python. So, you can add this prefix before create a new `Fxp` or when you get binary string, taking into account that you need to specify the fractional format. I can give you some examples to solve that:

```python
x1 = Fxp(3.4)
x_bin = x1.bin() # or '0b' + x1.bin()
x2 = Fxp('0b' + x_bin, like=x1) # also works with fractional dot format
```

If you don't want to keep `x1` you should create `Fxp` with a particular `dtype` or work with a _fractional binary string format_:

```python
x1 = Fxp(3.4)
x_bin = x1.bin(frac_dot=True)
x2 = Fxp('0b' + x_bin)
```

In the last example `x2` could have a different size than `x1` because leading or trailing zeros.

nfrancque, 2021-12-23:
Thanks for the quick response.  The first example works in my case since the original will still be around so I'll go with that. It'd be a nice enhancement to go from just the output string on its own (as long as `frac_dot` was specified) back to an `Fxp`.

francof2a, 2021-12-23:
Yes, it would be good!
I'll try to include that in next release. Thanks for the recommendation.

kokare-darshan, 2022-02-02:
Hi @francof2a , 

I am facing issue with fraction part using above method.
```
float_val = 256.2
ret_int = Fxp(float_val, n_word=32, signed=True, n_frac=2, n_int=29,)
ret_float = Fxp("0b"+ ret_int.bin(),  n_word=32, signed=True, n_frac=2, n_int=29,)
print(ret_float)
```

output is `256` but I need 256.2

francof2a, 2024-02-08:
solved in [v0.4.9](https://github.com/francof2a/fxpmath/releases/tag/v0.4.9)

closed 2024-02-08 by francof2a