francof2a/fxpmath #50: Docs:  insufficient explanation of raw value editing

jrmoserbaltimore, 2022-01-18:
The below does not do what I expect:

```python
a=Fxp(0xe25398fa9f, dtype='Q2.38')
a.set_val(0xe25398fa9f, raw=True)
```
The result is to set `a` to a positive value of all bits on (except the sine bit).

The result I expect is that `Fxp.bin() == 0b1110001001010011100110001111101010011111`

I've thus far been unable to determine from the docs what, if anything, would allow me to put in some figure and get that figure to be `Fxp.bin()`.  As you may notice, this binary string is a negative number.

francof2a, 2022-01-19:
Hello @jrmoserbaltimore 

In your example, there is a "problem" about how python represents hexadecimal values, because `0xe25398fa9f` is just an `int` (signed) equals to 972065143455 (not a negative number). When you create a `Fxp` with this value, its sign information was lost.

You have two options at least:

1. You can pass the hexadecimal value as a string:

```python
a = Fxp('0xe25398fa9f', dtype='Q2.38', raw=True)
# or
a.set_val('0xe25398fa9f', raw=True)

a.bin(frac_dot=True)
```

> '11.10001001010011100110001111101010011111'

2. Pass the value with sign information using _twos-complement_ (a function to do that is available in `fxpmath.utils`):

```python
from fxpmath.utils import twos_complement_repr

v = twos_complement_repr(0xe25398fa9f, nbits=40)
a = Fxp(v, dtype='Q2.38', raw=True)

a.bin(frac_dot=True)
```

> '11.10001001010011100110001111101010011111'

I hope this helps.

jrmoserbaltimore, 2022-01-20:
Thanks, that helps.  I may have not read the documents closely enough, I'll check again.  Didn't think to pass the number as a string.

closed 2022-02-16 by francof2a

javValverde, 2022-10-31:
Hello,

First of all congrats for the project! It's been a great discovery

I'm trying to understand why a string is needed when converting from integer back to fixed point. Given `dtype=Q2.38`, it is already known that the number is signed with 40 bits. If the MSB (i.e. bit 39) is one, then the number should be negative. Am I missing any other use case?

I can open a new issue if you prefer it