francof2a/fxpmath #52: Realistic expectations for numpy compatibility?

nfrancque, 2022-01-30:
Should we expect *all* numpy functions to work with fxpmath, or just the common ones?

I'm trying to call `numpy.linalg._umath_linalg.eigh_lo` and getting the following exception:

```
  File "{{site-packages}}fxpmath/objects.py", line 1577, in __array_ufunc__
    return self._wrapped_numpy_func(ufunc, *inputs, **kwargs)
  File "{{site-packages}}fxpmath/objects.py", line 1657, in _wrapped_numpy_func
    return self.__array_wrap__(val)
  File "{{site-packages}}fxpmath/objects.py", line 1126, in __array_wrap__
    return self.__class__(out_arr)
  File "{{site-packages}}fxpmath/objects.py", line 226, in __init__
    self._init_size(val, signed, n_word, n_frac, n_int, max_error=self.config.max_error, n_word_max=self.config.n_word_max, raw=raw)
  File "{{site-packages}}fxpmath/objects.py", line 357, in _init_size
    self.set_best_sizes(val, n_word, n_frac, max_error=max_error, n_word_max=n_word_max, raw=raw)
  File "{{site-packages}}fxpmath/objects.py", line 481, in set_best_sizes
    val, _, raw, signed, n_word, n_frac = self._format_inupt_val(val, return_sizes=True, raw=raw)
  File "{{site-packages}}fxpmath/objects.py", line 661, in _format_inupt_val
    val, signed, n_word, n_frac = utils.str2num(val, self.signed, self.n_word, self.n_frac, return_sizes=True)
  File "{{site-packages}}fxpmath/utils.py", line 154, in str2num
    x[idx], _signed, _n_word, _n_frac = str2num(v, signed, n_word, n_frac, base, return_sizes=True)
TypeError: 'tuple' object does not support item assignment
```

I tried messing with handling for tuples vs. lists a bit but wasn't able to figure much out.

francof2a, 2022-02-14:
_fxpmath_ works with _numpy_ through dispatching methods `__array_function__` and `__array_ufunc__` ([more info here](https://numpy.org/neps/nep-0018-array-function-protocol.html)) where a `Fxp` is converted to a `ndarray` before call the _numpy_ function and create a new `Fxp` with the returned value. If a numpy function doesn't support this protocol maybe an error could happen.

According to the error you reported, it seems that the problem is with the creation of a `Fxp` with the returned value. It would be useful to have a piece of code to reproduce the error and perform the debug.

francof2a, 2022-11-10:
Closing this issue due it's inactive.

closed 2022-11-10 by francof2a