francof2a/fxpmath #53: div for complex data does not behave expected

ruoting0103, 2022-02-11:
```
from fxpmath import Fxp
x = Fxp(2j, dtype = 'fxp-u4/0-complex')
x/2
```
the result is `fxp-u4/0-complex(0j)` rather than `1j`

i could solve the problem by adding  `x.config.op_method = 'repr'`

is there any other way to make it work like `2j/2` ? thanks

francof2a, 2022-02-14:
Clearly, it's a bug!

I did a debug and the problem is that a `floordiv` with a complex number doesn't work in _python_ nor _numpy_

```python
j2 // 2 # raise TypeError: can't take floor of complex number.

np.array(2j) // np.array(2) # result = 0j
```

The `raw` method is used bay default in Fxp operation to assure precision, when you use `repr` the operation is performed using a `truediv` (`/`).

In other way, if you use multiplication and give a fractional bit the operation is performed well, but I don't think this is a workaround:

```python
x = Fxp(2j, dtype = 'fxp-u4/1-complex')
x * 0.5
```
> fxp-u4/1-complex(1j)

A special method should be implemented for _complex division_ (raw op_method). It would be included in next release!

francof2a, 2022-04-01:
Issue solved in v0.4.6

closed 2022-04-01 by francof2a