francof2a/fxpmath #60: Incorrect Rounding Applied When resizing

DanChianucci, 2022-04-07:
There are certain situation when rounding isn't applied correctly.
Given the snippet below, you can see that rounding of the final two values in the array incorrectly rounds down.

It seems to have something to do with the dtype being used to store the value under the hood.  if t_fxp is changed such that the total number of bits <32, then the rounding gets applied correctly.  

I have also noticed that this issue is only occuring when the new array is created from an existing FXP array.  ie using .resize, or initializing using another FXP array as the initial value.   The rounding is correct if initializing from an array of floats.
It also looks like the rounding is correct for FXP initialized from a scalar float.

```python
from fxpmath import Fxp, Config
import numpy as np

cfg=Config(dtype_notation="Q",rounding="around")
t_fxp = Fxp(0.0,1,n_int=16,n_frac=15,config=cfg)   # Rounding fails if sign+n_int+n_frac >= 32
t_int = Fxp(0.0,1,n_int=13,n_frac=0,config=cfg)

arr = [-5,0,14.8,7961.625]
fullprec        = Fxp(arr      , like=t_fxp )
rounded_direct  = Fxp(arr      , like=t_int )    # Works
rounded         = Fxp(fullprec , like=t_int )    # Doesn't Work
print(repr(fullprec))
print(repr(rounded_direct))
print(repr(rounded))

scalar_full         = Fxp(7961.625     , like=t_fxp )
scalar_round_direct = Fxp(7961.625     , like=t_int )            #<- Works
scalar_round        = Fxp(scalar_full  , like=t_int )            #<- Also Works
print(repr(scalar_full))
print(repr(scalar_round_direct))
print(repr(scalar_round))

```

The above will print out

Q17.15([-5.00000000e+00  0.00000000e+00  1.47999878e+01  7.96162500e+03])
Q14.0([-5.000e+00  0.000e+00  1.500e+01  7.962e+03])
Q14.0([-5.000e+00  0.000e+00  1.400e+01  7.961e+03])

Q17.15(7961.625)
Q14.0(7962.0)
Q14.0(7962.0)

francof2a, 2022-04-09:
Hello @DanChianucci 
Thanks for report the issue. Which version are you using?

I ran your example with v0.4.6 and it is working.

francof2a, 2022-04-12:
You're right! I tested it in windows and I could replicate the problem.

When `fullprec` is created, a internal raw value has 32 bits. In this case, _numpy_ rounding functions don't work, so rounding process is by-passed (maybe I should add a _warning message_), although even, precision of `float` (32 bits) make you lose information to be rounded.
How the value passed to `rounded` doesn't have a resolution greater than 32 bits, it could be fixed internally (I will include that in next release).

Just like a workaround (well... here **work** **around** is not a "happy" word, hehe), you can try:

```python
rounded         = Fxp(fullprec() , like=t_int ) #note () after fullprec to pass float values instead Fxp
```

francof2a, 2022-04-15:
Please, try new version 0.4.7 and check if now it is working in Windows.

francof2a, 2022-04-21:
I have already test in a PC with Windows 64 bits, and could replicate your conditions (`uint` equivalent to `uint32`). The problem was a mistake of mine to not include a change from dev branch.
The comparison in lines you mentioned are necessary to deal with this problem, forcing to work numpy with python objects instead of `uint`. The problem I found is that when input value has raw value as `object` type.

You should get it fixed doing including following code line between line 809 and 810 of _objects.py_:

```python
val = val.astype(original_vdtype)
```
this results in:

```python
            else:
                val = val.astype(original_vdtype) # <----- added line
                val_dtype = np.int64 if self.signed else np.uint64
```

If you see line 840 to 842, this change is already implemented for complex values. 
**Please, let me know if you try this!**

Version 0.4.7 should have this change :( !!
I've run test again, and it pass ok!

```python
def test_issue_60_v0_4_6():
    cfg=Config(dtype_notation="Q",rounding="around")

    t_fxp = Fxp(0.0,1,n_int=16,n_frac=15,config=cfg)
    t_int = Fxp(0.0,1,n_int=13,n_frac=0,config=cfg)

    arr = [-5,0,14.8,7961.625]
    fullprec        = Fxp(arr      , like=t_fxp )
    rounded_direct  = Fxp(arr      , like=t_int )
    rounded         = Fxp(fullprec , like=t_int )
    assert np.all(rounded_direct == rounded)

    scalar_full         = Fxp(7961.625     , like=t_fxp )
    scalar_round_direct = Fxp(7961.625     , like=t_int )
    scalar_round        = Fxp(scalar_full  , like=t_int )

    assert scalar_round_direct == scalar_round
```

DanChianucci, 2022-04-21:
Yep! That seems to have fixed the issue

francof2a, 2022-04-24:
I close the issue considering it solved in version 0.4.8.
Thanks!

closed 2022-04-24 by francof2a