francof2a/fxpmath #66: Unexpected behavior when using equal() with a numpy array

pablokrupa, 2022-06-02:
The following code results in an unexpected behavior. At least for me.
```
x = Fxp( np.array([1.25, 0.5]), dtype='S8.4' )
y = Fxp( np.array([2.25, 1.5]), dtype='S8.4' )
x[0].equal(y[0])
print(x[0])
```
I expected it to print 2.25, but it prints 1.25.
The assignment works it you use x[0] = y[0]. It also works if you don't use arrays, i.e.,
```
x = Fxp(1.75, dtype='S8.4')
y = Fxp(2.75, dtype='S8.4')
x.equal(y)
print(x)
```
prints 2.25. This issue seems related to #62.

francof2a, 2022-06-03:
You're right about this issue is related to #62.

When you do ```x[0]```, you get a new `Fxp` with a copy of the indexed raw value (numpy does not create a view in this case). So, when you use `equal` method, you are calling the method of the new `Fxp` and not from original `Fxp` object.

This could be solved using `x[0] = y[0]` how you mentioned.

`equal` should be use when you need to set equal values (not raw values) between two `Fxp` objects with different sizes. 

According this issue, an _index_ parameter could be added to `equal`, where an example of use could be:

```python
x.equal(y[0], index=0)
```

pablokrupa, 2022-06-04:
Thanks for the answer.

I was using `equal` to ensure that the fractional and integer bits did not change. I posted a minimum example, but in my case it wasn't as simple as `x[0] = y[0]`, but a more complex assignment in which I wanted x to preserve its precision. Say `x[0] = y[0] * z[0] - y[2]`, just as an invented example.

The thing that confused me at first is that `x = y` changes the fractional precision of `x` if `x` and `y` are scalars (e.g., `x = Fxp( np.array([1.25, 0.5]), dtype='S8.3' )` ), so if I want to force `x` to maintain its fractional precision then I should use `x.equal(y)`. However, when dealing with numpy arrays its different. Now `x[0] = y[0]` does not change the fractional bits of `x` (even if `y` has higher precision), and `x[0].equal(y[0])` does not update `x[0]`, as already discussed. However, `x.equal(y)` behaves as in the scalar case (I can send you some minimal examples of these cases if you want, but I think it is quite clear).

The issue here is not the behavior of `x[0].equal(y[0])`. The reason why it behaves as it does is quite clear from your explanation and very reasonable. The issue is that this may not be immediate to the user due to the non-intuitive behavior difference between scalars and arrays. Therefore, I think your suggestion of the index parameter is very good. That plus maybe adding an example of this in the README.md so that the user is aware of it.

I'm happy to help here if need be.

francof2a, 2024-02-08:
solved in [v0.4.9](https://github.com/francof2a/fxpmath/releases/tag/v0.4.9)

closed 2024-02-08 by francof2a