francof2a/fxpmath #67: Complex "dtype" error happen when certain fractional precision

VincentChen1017, 2022-06-09:
Hi, I'm trying to use fixed point to simulate the FFT and generate the golden pattern for hardware design.

I set my input signal to be 'fxp-s32/23', and to avoid the possible overflow situation, so I try to large the temp variable's size.

When I set the temp variable to be 'fxp-s65/46', it occur ERROR message like : AttributeError: 'complex' object has no attribute 'dtype'.

But if I test the complex operation with 'fxp-s65/46 using other simple equation, or change the  temp variable in my fft to be 'fxp-s65/23' (or reduce the n_frac part to certain value), then it will work without error.

I'm new to the fxpmath library, I tried hard to solve it but still not work :(
Following is my  code, is there any wrong operation with the "Fxp" ?  Thanks !

    input_size = Fxp(None, dtype='fxp-s32/23')
    f = [0,10+7j,20-0.65j,30]
    f = Fxp(f, like = input_size)

    def FFT(f):
    N = len(f)
    if N <= 1:
        return f
    
    # division: decompose N point FFT into N/2 point FFT
    even= FFT(f[0::2])
    odd = FFT(f[1::2])

    # store combination of results
    temp = np.zeros(N, dtype=complex)
    temp = Fxp(temp, dtype='fxp-s64/23')

    for u in range(N//2):
        W =  Fxp(exp(-2j*pi*u/N), like=input_size) 
        temp[u] = even[u] + W* odd[u] 
        temp[u+N//2] = even[u] - W*odd[u]  
        
    return temp

    # testing the function to see if it matches the manual computation
    F_fft = FFT(f)
    print(F_fft)

VincentChen1017, 2022-06-15:
Hello, 

when I ran code like this, it's OK.

    temp = Fxp(temp, dtype='fxp-s65/23')

but I change n_frac to 46 like:

    temp = Fxp(temp, dtype='fxp-s65/46')

It's error report in the following:

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-10-5c240a2ee42d> in <module>
     24 
     25 # testing the function to see if it matches the manual computation
---> 26 F_fft = FFT(f)
     27 print(F_fft)

<ipython-input-10-5c240a2ee42d> in FFT(f)
      9 
     10     # division: decompose N point FFT into N/2 point FFT
---> 11     even= FFT(f[0::2])
     12     odd = FFT(f[1::2])
     13 

<ipython-input-10-5c240a2ee42d> in FFT(f)
     18     for u in range(N//2):
     19         W =  Fxp(exp(-2j*pi*u/N), like=input_size)
---> 20         temp[u] = even[u] + W* odd[u]
     21         temp[u+N//2] = even[u] - W*odd[u]
     22 

~/opt/anaconda3/lib/python3.7/site-packages/fxpmath/objects.py in __mul__(self, x)
   1223             _sizing = self.config.op_sizing
   1224 
-> 1225         return mul(self, x, out=self.config.op_out, out_like=self.config.op_out_like, sizing=_sizing, method=self.config.op_method)
   1226 
   1227     __rmul__ = __mul__

~/opt/anaconda3/lib/python3.7/site-packages/fxpmath/functions.py in mul(x, y, out, out_like, sizing, method, **kwargs)
    362     optimal_size = (signed, n_word, n_int, n_frac)
    363 
--> 364     return _function_over_two_vars(repr_func=np.multiply, raw_func=_mul_raw, x=x, y=y, out=out, out_like=out_like, sizing=sizing, method=method, optimal_size=optimal_size, **kwargs)
    365 
    366 @implements(np.floor_divide)

~/opt/anaconda3/lib/python3.7/site-packages/fxpmath/functions.py in _function_over_two_vars(repr_func, raw_func, x, y, out, out_like, sizing, method, optimal_size, **kwargs)
    176         z = out.set_val(val, raw=raw)
    177     else:
--> 178         z = Fxp(val, signed=signed, n_int=n_int, n_frac=n_frac, like=out_like, raw=raw, config=config)
    179 
    180     return z

~/opt/anaconda3/lib/python3.7/site-packages/fxpmath/objects.py in __init__(self, val, signed, n_word, n_frac, n_int, like, dtype, **kwargs)
    235 
    236         # store the value
--> 237         self.set_val(val, raw=raw)
    238 
    239     # ---

~/opt/anaconda3/lib/python3.7/site-packages/fxpmath/objects.py in set_val(self, val, raw, vdtype, index)
    870 
    871         # update dtype
--> 872         self._update_dtype()
    873 
    874         # vdtype

~/opt/anaconda3/lib/python3.7/site-packages/fxpmath/objects.py in _update_dtype(self, notation)
    751                                                                     nword=self.n_word,
    752                                                                     nfrac=self.n_frac,
--> 753                                                                     comp='-complex' if (self.val.dtype == complex or self.vdtype == complex) else '')
    754             else:
    755                 self._dtype = 'fxp-{sign}{nword}/{nfrac}'.format(sign='s' if self.signed else 'u', 

AttributeError: 'complex' object has no attribute 'dtype'

---------------------------------------------------------------------------

My OS version is macOS 10.13.6, fxpmath version is 0.4.8, python version is 3.7.6, numpy version is 1.18.6.

If need more information just let me know. Thank you !

francof2a, 2024-02-08:
solved in [v0.4.9](https://github.com/francof2a/fxpmath/releases/tag/v0.4.9)

closed 2024-02-08 by francof2a