francof2a/fxpmath #73: underflow / saturation of unsigned list not working

AndrzejKowalski9917, 2022-10-26:
## Description
When subtracting a larger unsigned number from a smaller one the result is correctly 0.0. If both values are part of list, the result is wrong, if an underflow occurs (seems to be set to the upper limit, instead of the lower one).

## Minimal example:
```python
from fxpmath import Fxp

# single unsigned value does work
a = Fxp(10, False, 14, 3)
b = Fxp(15, False, 14, 3)
c = a - b
print(c)  # 0.0 --> correct
c.info(verbose=3)

# unsigned list does not work
d = Fxp([10, 21], False, 14, 3)
e = Fxp([15, 15], False, 14, 3)
f = d - e
print(f)  # [4095.875 6.0] --> 4095.875 is the upper limit
f.info(verbose=3)
```

`c.info(verbose=3)` delivers the following output:
> dtype           =       fxp-u15/3
    Value           =       0.0
    underflow       =       True
    inaccuracy      =       True
    Signed          =       False
    Word bits       =       15
    Fract bits      =       3
    Int bits        =       12
    Val data type   =       None
    Upper           =       4095.875
    Lower           =       0.0
    Precision       =       0.125
    Overflow        =       saturate
    Rounding        =       trunc
    Shifting        =       expand

`f.info(verbose=3)` this output:
> dtype           =       fxp-u15/3
   Value           =       [4095.875 6.0]
   overflow        =       True
   inaccuracy      =       True
   Signed          =       False
   Word bits       =       15
   Fract bits      =       3
   Int bits        =       12
   Val data type   =       None
   Upper           =       4095.875
   Lower           =       0.0
   Precision       =       0.125
   Overflow        =       saturate
   Rounding        =       trunc
   Shifting        =       expand

## Version infos
* OS:          Windows 10 21H2 (19044.2130)
* python:   3.7.6
* fxpmath: 0.4.8

francof2a, 2022-11-10:
Hello @AndrzejKowalski9917 

I tested your example in a fresh conda environment with python 3.7.6 (conda), fxpmath 0.4.8 (from anaconda) and I couldn't reproduce de error.
I also tested in a conda env with python 3.8.13 and fxpmath installed from Pip, and it's also working well.
Both runs were performed in linux, so I can not tell about relationship with Windows OS yet.

What version of _numpy_ are you using?
Could you run again your example in a fresh conda env?

AndrzejKowalski9917, 2022-11-10:
Hi @francof2a ,

thank you for your answer. I tried now in Linux (Arch) with
* python 3.10.8
* numpy 1.23.4
* 0.4.8

and also were not able to reproduce the error.  In Windows I also tried python 3.10.0, installed via the installer from python.org, fxpmath and numpy installed via pip (same versions as on Linux). Here the error occurs.
I can try a fresh conda env in Windows tomorrow at work.

AndrzejKowalski9917, 2022-11-11:
Hello @francof2a,

I just tried at work again (Windows 10) with a clean virtualenv (conda env does not work here for reasons unknown).
Python is still version 3.7.6 and only the following packages are installed:

> (test_env) D:\tmp\py>pip list
   Package    Version
    ---------- -------
   fxpmath    0.4.8
   numpy      1.21.6
   pip        22.3.1
   setuptools 65.5.1
   wheel      0.38.4

The error persists. So it seems to be an error specific to the Windows OS.

commit 85d731cc (2024-02-10): fix numpy int size for windows OS.
This fix #73 #76 and #85 in windows OS using numpy 32 bit integer type