francof2a/fxpmath #74: Performance

nfrancque, 2022-11-08:
This library is very useful but very very slow.  What can be done to improve performance?  I would like to keep using this because of the over/underflow tracking, but need to do fixed point in real-ish-time on large images.

francof2a, 2022-11-08:
Yes, you're right. The performance is a issue, but being implemented entirely in python and support integration with numpy, it's difficult to improve the performance (_foot note: the performance was not a main goal_). However, there are some 'good practices' when using the library that improve the performance.
Could you post a piece of code of an example about what you need? Maybe I can realize a way to improve its performance a little bit.

nfrancque, 2022-11-08:
I can't post the actual code, but it's all image processing based, so generally 1920x1080 numpy arrays of somewhere between 8 and 32 bit depth, and doing some fixed point math on them.  I'm doing the math in loops rather than using numpy operators, which I understand is probably not the most performant way to do it.  Currently I have a 600x200x24-bit frame that takes about 5 minutes per frame.

Something like:

```python
for y in range(height):
  for x in range(width):
    b, g, r, = image[y,x]
    something = b*whatever1 + r*whatever2
    output[y,x] = something

```

francof2a, 2022-11-08:
There are some considerations that could be important to mention considering this code (knowing that is not your actual code, but I hope you can infer some improvement):

1. It's really important take advantage from numpy operation instead for loops in order to improve performance. When you execute a only array operation instead a element-wise For-Loop operation, Fxpmath can take advantage of numpy performance to perform calculation with same data precision.
2. Fxpmath takes some relevant time to initialize new Fxp objects in explicit (i.e. variable `something` or `b`) or implicit (i.e. `b*whatever1`), so try to avoid this if you are going to repeat too many times.
3. If you are operating with constants, define them as Fxp objects also, using the precision you need. This will prevent that Fxpmath estimate the best size for that constant before execute the operation (i.e. `whatever1` in `b*whatever1`).
4. In some cases is important to change the precision of resulting size of operations. For example, if you do `var1*var2`, the resulting size is the sum of the bits of both variables, and maybe you don't need to increase your precision.

I did some runs:

## Numpy and For

```python
import numpy as np
from fxpmath import Fxp
from datetime import datetime

bits_depth = 24
height = 60
width = 20

image = np.random.randint(0, 2**bits_depth - 1, size=(height,width,3))
output = np.zeros((height,width))
whatever1 = 1.5
whatever2 = 0.3

startTime = datetime.now()
for y in range(height):
  for x in range(width):
    b, g, r, = image[y,x]
    something = b*whatever1 + r*whatever2
    output[y,x] = something
print(datetime.now() - startTime)
``` 
> Execution time: 0:00:00.005486

## Fxpmath and For

```python
image = np.random.randint(0, 2**bits_depth - 1, size=(height,width,3))
image = Fxp(image, dtype='U24.0')
output = np.zeros((height,width))
output = Fxp(output, dtype='U24.0')
whatever1 = Fxp(1.5, dtype='U24.0')
whatever2 = Fxp(0.3, dtype='U24.0')

startTime = datetime.now()
for y in range(height):
  for x in range(width):
    b, g, r, = image[y,x]
    something = b*whatever1 + r*whatever2
    output[y,x] = something
print(datetime.now() - startTime)
``` 

> Execution time: 0:00:05.519634

## Little improvement

You can get a little improvement according point 2 above:

```python
for y in range(height):
  for x in range(width):
    output[y,x] = image[y,x,0]*whatever1 + image[y,x,2]*whatever2
```

## Numpy optimized

```python
startTime = datetime.now()
output2 = image[:,:,0]*whatever1 + image[:,:,2]*whatever2
print(datetime.now() - startTime)
```

> Execution time: 0:00:00.000050

## Fxpmath optimized

Same code as **Numpy optimized** but using Fxp variables.

> Execution time: 0:00:00.008062

## Conclusion

You can see that there is a huge improvement between _For-Loop_ and _array operation_. Obviously, there will be always a big difference between execution time between _numpy_ and _fxpmath_, but in your case this will be relevant to use array operations.

Using **optimized operation** with _600x200x3 (24 bits)_ size image I got:

> Execution time: 0:00:00.001768 (numpy)
> Execution time: 0:00:00.493377 (fxpmath)

I think a half second it a good time in comparison with several minutes!

deere-nathanfrancque, 2022-11-10:
Thank you! I will see if I can convert my code over to array operations.

Why do you think it is that there is such a mismatch between numpy for and fxpmath for?

nfrancque, 2022-11-10:
Thank you! I will see if I can convert my code over to array operations.

Why do you think it is that there is such a mismatch between numpy for and fxpmath for?

francof2a, 2022-11-10:
You're welcome. Thanks for the feedback!

When the code makes operation separately (in your for-loop for example) there are some process to creation, validation and conversions from `float` to `fxp` and viceversa which are implemented in python (not in C/C++ as numpy engine). 
Those processes make a performance reduction against a pure _numpy_ operations. When you execute array functions, those process I mentioned are executed only once, and then all calculation is thrown to _numpy_.

Right now, _fxpmath_ prioritizes functionality over performance, but it would be better improve that!

Do you consider this issue ready to be closed?

closed 2023-12-20 by francof2a