francof2a/fxpmath #80: Creation of Fxp-object with negative n_frac

alb-pol, 2023-03-23:
The following code results in unexpected behaviour when trying to specify the same type using alternative formats :

`
x = Fxp(16, signed=True, n_word=8, n_frac=-2) 
`
-> x.dtype = 'fxp-s8/-2' , ok

`
x = Fxp(16, dtype='S10.-2')
`
-> x.dtype = 'fxp-s10/0' ,  why? expected x.dtype= 'fxp-s8/-2' 

`
x = Fxp(16, dtype='fxp-s8/-2')
`
-> error message

> Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "c:\Projects\MA-pyrdf\.venv\lib\site-packages\fxpmath\objects.py", line 222, in __init__
    signed, n_word, n_frac, complex_flag = self._parseformatstr(dtype)
  File "c:\Projects\MA-pyrdf\.venv\lib\site-packages\fxpmath\objects.py", line 348, in _parseformatstr
    raise ValueError('unrecognized format string')
ValueError: unrecognized format string

francof2a, 2024-01-19:
Hello,
I could replicate the bug. The problem is that the _dtype string parsing_ does not support negative numbers.
This will be fixed in next version (v0.4.9).
Thank you for the issue report!

francof2a, 2024-02-08:
solved in [v0.4.9](https://github.com/francof2a/fxpmath/releases/tag/v0.4.9)

closed 2024-02-08 by francof2a