francof2a/fxpmath #87: Feature Request: Prevent auto-scaling

Saad-Mufti, 2023-10-27:
This library is really comprehensive, and does a great job of abstracting away whatever types Python has to make it feel like a native fixed-point type truly exists.

I think one very useful feature would be having the ability to disable autoscaling, especially if you have a use-case where you are building a system that has fixed-point types, and you want to develop a behavioral model on Python to see play around with these types.

I was hoping this could be interfaced like so:
```
x = fixed(12.2).like(some_format, autoscale=False)
```

danilo-bc, 2023-11-02:
I think I use the library for a similar use case, so I think this feature is already present. Can you give an example?

For example, if I use
```python
 from fxpmath import Fxp
>>> a = Fxp(0.5, n_word=8, n_frac=0)
fxp-s8/0(0.0)
>>> b = Fxp(0.5, n_word=8, n_frac=1)
fxp-s8/1(0.5)
```
This is exactly as intended. The decimal number can be represented in "b", but not in "a". And since you want to develop a behavioral model with fixed-point types, you probably want numbers to be automatically "scaled".

If you really want to "disable autoscaling", just take away the "like". `Fxp(12.2)` will try to represent that number in fixed-point notation using as few bits as possible.

francof2a, 2023-12-20:
This functionality is already implemented and is the **template**, if I understood well what are your expectations.

You can see some details at [v0_4_0_new_functionalities](https://github.com/francof2a/fxpmath/blob/master/examples/v0_4_0_new_functionalities.ipynb) notebook (_template_ section). There is no information about it in the documentation yet!

```python
# Template is defined inside Fxp class
Fxp.template = Fxp(None, dtype='fxp-s16/8')

x = Fxp(12.2)
x.info()
```
> dtype           =       fxp-s16/8
> Value           =       12.19921875
> inaccuracy      =       True

```python
y = x * 4096
y.info()
```
> dtype           =       fxp-s16/8
> Value           =       127.99609375
> overflow        =       True
> inaccuracy      =       True

Check that resulting size is not scaled in order to keep optimal size to store accurate result.

closed 2024-02-08 by francof2a