francof2a/fxpmath #89: Infinite loop on exponent somehow

jrmoserbaltimore, 2023-12-03:
I don't know what particularly causes this, but it hangs forever.

```python
from fxpmath import Fxp
x = Fxp(0,dtype='UQ0.64')
x**4
```

francof2a, 2023-12-20:
That is because when you use a constant as is, in this case the number 4, as operator of a `Fxp` object, this constant is converted to an `Fxp` object with same _dtype_. Then, the power operation are using a huge raw number as exponent.

You can solve this being explicit about dtype of the exponent:

```python
x**Fxp(4)
# or
x**Fxp(4, dtype='some fixed-point dtype')
```

jrmoserbaltimore, 2023-12-20:
Interesting.  Should this produce a warning or raise an exception (probably the best route) or be a different behavior, or is that just impossible?  It's not obvious what's happening and an infinite loop is probably not the programmer's intent.

francof2a, 2023-12-22:
Yes, it could be! I think a good change can be modify the `op_input_size` default value from `'same'` to `'best'`. In the first case the number 4 is converted to `UQ0.64` before to perform the power operation. In the second case, using `'best'` the number for is converted to `Q4.0`. This behavior will prevent long calculation time and intensive memory usage.

But the idea of the **warning** is really important in order be aware of this automatic sizing. Definitely I'll include this in next release.

francof2a, 2024-02-08:
solved in [v0.4.9](https://github.com/francof2a/fxpmath/releases/tag/v0.4.9)

closed 2024-02-08 by francof2a