francof2a/fxpmath #90: Bug: Built-in `Fxp.reshape()` method passes the wrong number of arguments to underlying numpy `reshape()` function.

Jonah-Foley, 2024-02-01:
# Recreation of the issue

```
x = Fxp([1,2,3,4], False, 8, 0)
x = x.reshape((2,2))
```
Error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/.../.venv/lib64/python3.9/site-packages/fxpmath/objects.py", line 587, in reshape
    self.val = self.val.reshape(shape=shape, order=order)
TypeError: function takes at most 1 keyword argument (2 given)
```

# Proposed solution

We need to re-implement the `Fxp.reshape()` method as 

```
self.val = self.val.reshape(shape, order=order)
```

francof2a, 2024-02-10:
Hi, I could reproduce the issue.
I modified the code using your solution but also modify the method behavior to match _numpy_ behavior not modifying its self `val` shape. Now, it only return a reshaped `Fxp` object.

This will be fixed in version `0.4.10`.

Thanks.

Jonah-Foley, 2024-02-12:
Great to hear, thank you.

closed 2024-02-12 by Jonah-Foley