francof2a/fxpmath #97: Incorrect Multiplication Result when using NumPy Array Instantiation

Greek64, 2024-06-20:
I'm getting wrong results when I'm using Fxps instantiated from a NumPy Array and given a custom dtype.
The same Fxp instantiated via different ways, even if seemingly exactly the same, produces correct results.

---

Minimal Example:
```
import numpy as np
from fxpmath import Fxp

a1 = Fxp(np.array([1.057311]),dtype='Q5.20')
print(f"A1 = {a1[0]}, dtype = {a1[0].dtype}, mem = {a1[0].val}")
a2 = Fxp(1.057311,dtype='Q5.20')
print(f"A2 = {a2}, dtype = {a2.dtype}, mem = {a2.val}")
b = Fxp('7971079',dtype='Q24.20')
print(f"B = {b}, dtype = {b.dtype}")
c = b * a1[0]
print(f"C = {c}, dtype = {c.dtype}")
d = b * a2
print(f"D = {d}, dtype = {d.dtype}")
e = b * Fxp(a1[0],dtype='Q5.20')
print(f"E = {e}, dtype = {e.dtype}")
```

Output:
```
A1 = 1.0573101043701172, dtype = fxp-s25/20, mem = 1108670
A2 = 1.0573101043701172, dtype = fxp-s25/20, mem = 1108670
B = 7971079.0, dtype = fxp-s44/20
C = -8349313.630567551, dtype = fxp-s69/40
D = 8427902.36943245, dtype = fxp-s69/40
E = 8427902.36943245, dtype = fxp-s69/40
```

As can be seen from the output, Fxp `A1` produces incorrect multiplication results.
I don't know if it affects different operations besides multiplication and other conditions that have to be met.

francof2a, 2026-03-12:
I reproduced the bug from #97 with the reported case:

- `A1 = Fxp(np.array([1.057311]), dtype='Q5.20')`
- `A2 = Fxp(1.057311, dtype='Q5.20')`
- `B = Fxp('7971079', dtype='Q24.20')`

`B * A1[0]` produced an incorrect negative result, while `B * A2` and `B * Fxp(A1[0], dtype='Q5.20')` matched and were correct.

### Root Cause

The issue is a scalar-type mismatch in the raw/object arithmetic path:

- `__getitem__` can return a scalar-backed `Fxp` where `val` is a `numpy.int64` scalar.
- In object-cast mode, `np.array(x, dtype=object)` preserved that as a NumPy scalar object.
- Multiplication then used NumPy fixed-width integer semantics and overflowed (instead of Python arbitrary-precision integer behavior).

So this was not a sizing/dtype metadata issue; it was a NumPy scalar vs Python scalar behavior issue in the overflow-safe path.

### Fix Implemented

Updated `_cast_to_object` in `fxpmath/helpers.py` to normalize NumPy scalars before object-casting:

- If input is `np.generic`, convert with `.item()` first.
- Then cast with `np.asarray(x).astype(object)`.

This guarantees Python scalar objects in object-mode arithmetic, preventing fixed-width overflow in this path.

### Validation

- Added regression test `test_issue_97_v0_4_9` in `tests/test_issues.py` asserting indexed/scalar multiplication consistency.
- Ran full test suite: **98 passed, 0 failed** (2 pre-existing warnings).

This will be included in `v0.5.0`

francof2a, 2026-03-24:
Solved in version `0.4.10`